bastianraschke/pyfingerprint #1: Unable to initialize Fingerprint sensor

gauravdave01, 2016-05-14:
I am using this library to connect R305 (ZFM60_BAK_V1.4). When I try to enroll a new finger, it gives this error:
```
The fingerprint sensor could not be initialized!
Exception message: The received packet do not begin with a valid header!
```

Instead of `/dev/ttyUSB0`, I'm doing `/dev/ttyAMA0` and Baud rate is: 9600 (I've tried with 115200 and 57600). But nothing works.

bastianraschke, 2016-05-17:
Hey gauravdave01, you tried to exchange the data wires? Or used another computer, USB serial converter? Often this is the problem.

gauravdave01, 2016-05-18:
Actually, I'm connecting the sensor with female-to-female jumper wires and then making a connection.

Mehluko, 2016-11-20:
Did you find a way to fix this problem?

bastianraschke, 2016-11-21:
Hey Mehluko, this is a really common problem. Mostly it is caused by bad wires or bad USB serial converter or baudrate. Anything else its hard to guess.

gauravdave01, 2016-11-24:
Hey @Mehluko,

Please find steps which I've executed for integration:

So, I haven't used USB connector to make connection, I directly connected Fingerprint device with Raspeberry Pi 2's (Model B) GPIO.

This is how I have connected fingerprint device:

```
Black Wire ---- GPIO 6 [Ground]
White Wire ---- GPIO 15 [RX]
Green Wire ---- GPIO 8 [TX]
Red Wire ---- GPIO 1 [3V PWR]
```

I've tried these connection as well:
- Red wire to GPIO 2 [again 5V].
- Interchange RX connection with TX and vice versa.

I've made all the required basic changes, e.g. replaced ttyUSB0 with ttyAMA0 and other changes as described.

Still, I was getting error, then I tried with 52600 baud rate and it started _working!_.

I hope this could help.

rubiagatra, 2017-03-04:
hello @gauravdave01, when I tried with 52600 I had this output

`stty: invalid argument ‘52600’
Try 'stty --help' for more information.
`

Rajashri1234, 2017-03-11:
hi, I am getting same error 
The fingerprint sensor could not be initialized! 
Exception Message: The received packet do not begin with a valid header!

Did anyone find solution for this ?
I tried changing Tx & Rx connection. But it did not help
I am also making connections same as that of @gauravdave01 . Directly connecting Pi with sensor using female to male connection.

EsEm17, 2017-03-24:
We have encountered this error:

Operation failed!
Exception message: storeTemplate() takes at least 2 arguments (1given)

Can you help us out? Pleease. Thank you.

gauravdave01, 2017-03-24:
Hey @rubiagatra & @Rajashri1234,

Try running this command:

`@reboot sudo stty -F /dev/ttyAMA0 52600`

I think, this will change the default baudrate of **115200** to **52600**. Then try again with the connection I've describe before.

See, if that works.

philippmeisberger, 2017-03-24:
@EsEm17 please do not post your problem neither on closed nor opened issues that have nothing to do with it. If you tried your best and nothing helped open a new issue instead. Thank you

EsEm17, 2017-03-27:
@philippmeisberger I don't know how okay I'll find ways. Thank you. I'm so sorry.

closed 2017-04-16 by bastianraschke

brianmak3, 2018-01-20:
Hello, i manged to solve this problem by first changing the baud rate to 57600 as for my pi using this command line**sudo stty -F /dev/ttyAMA0 52600** then i interchanged the wires and it worked.

Ashwinigunjal, 2018-07-30:
I am using  serial /dev/ttyS0 for communincation. it send packet successfully. but return invalid header. why this happen.

Besomi996, 2018-12-23:
hello i tried the things that you proposed in here non has worked i thought maybe because i have FS-01 fingerprint sensor and i am using dev/ttyUSB0 please help me i cant make it work it keeps giving me the same error
The fingerprint sensor could not initialized
and i have followed all of the steps proposed here

philippmeisberger, 2018-12-24:
Never heard about FS-01. Can you provide a datasheet? Probably the FS-01 is not compatible with PyFingerprint.

Besomi996, 2018-12-24:
@philippmeisberger  yes of course you can find the manual here:
http://www.fingergo.cn/uploads/soft/FS-01_Fingerprint_Sensor_Manual_Download.pdf
I thought it is the problem but if doesnt work for these libraries then how does it work ?

philippmeisberger, 2018-12-25:
I checked the datasheet: I am sorry but this library does not support the FS-01

philippmeisberger, 2018-12-27:
Please take a look at #44 and the FAQ (https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/)

cjoshi015, 2019-05-06:
I have also encountered this problems

use this pin numbering

1 | 5V | Regulated 5V DC
2 | GND | Common Ground
3 | TXD | Data output - Connect to MCU RX
4 | RXD | Data Input - Connect to MCU TX
5 | TOUCH | Active Low output when there is touch on sensor by finger
6 | 3.3V | Use this wire to give 3.3V to sensor instead of 5V

and also check your jumber cables are connected properly or not majorly it is due to loose connection and on many sited pin number is given wrong use above pin numbers

I have solved my issue using above steps