bastianraschke/pyfingerprint #2: Template count

motxo88, 2016-07-21:
Hello bastian, i'm using your code and i have noticed, when i enroll multiple fingers it doesnt increment the position stored. It always stores at ID #1.

bastianraschke, 2016-07-22:
Hey motxo88, in which example you noticed that behavior?

motxo88, 2016-07-22:
well, i tested out just now: 

when i enrolled different fingers the first time using the example_enroll, it always stored the template at position 1.

then i used the example_delete, to delete the only template stored (position 1), and now when i enroll multiple fingers, it count's right and stores right, beggining to store at position 0 and so on. 

but if i delete, for example the template stored at the position 0, it goes back to the same, it stores the next templates, at the last template position and doesn't increment anymore, but if i delete the last template stored, it all goes well.

these we're the tests that i made.

bastianraschke, 2016-07-31:
Hm thats strange, maybe the template count works not properbly on your sensor? When you use the enroll example, the output "Currently stored templates:" changes when a finger has been enrolled?

philippmeisberger, 2016-08-05:
This is because of lines 69 in example_enroll.py:
...
positionNumber = f.getTemplateCount()
...

When 3 templates are enrolled and the 2. is deleted the count is 2. When another template is enrolled this will be stored at index 2 and overwrite the 3. template from before deletion.

bastianraschke, 2016-09-27:
Will be fixed in #5.

closed 2016-09-27 by bastianraschke

EsEm17, 2017-03-24:
We have encountered this error:
Operation failed!
Exception message: storeTemplate() takes at least 2 arguments (1given)
Can you help us out? Pleease. Thank you.