bastianraschke/pyfingerprint #8: Unable to debuild

rubiagatra, 2017-02-28:
I can't debuild the src and it has output like this 

` dpkg-buildpackage -rfakeroot -D -us -uc
dpkg-buildpackage: source package python-fingerprint
dpkg-buildpackage: source version 1.4
dpkg-buildpackage: source distribution stable
dpkg-buildpackage: source changed by Philipp Meisberger <team@pm-codeworks.de>
 dpkg-source --before-build src
dpkg-buildpackage: host architecture amd64
dpkg-checkbuilddeps: error: Unmet build dependencies: debhelper (>= 9)
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
dpkg-buildpackage: warning: (Use -d flag to override.)
debuild: fatal error at line 1376:
dpkg-buildpackage -rfakeroot -D -us -uc failed`

what kind of dependencies I should install?

philippmeisberger, 2017-02-28:
Which OS are you using?

rubiagatra, 2017-03-01:
I am using Ubuntu 16.04.2 . Does it works with ubuntu?

philippmeisberger, 2017-03-01:
Of course it works on Ubuntu. The second error you pasted is usual as debsign tries to sign the package with a the GPG key of the maintainer. 

Nice to know that `dpkg-buildpackage -uc -us` is the correct command for building the package as non-maintainer. Bastian should mention that in the readme ;)

rubiagatra, 2017-03-02:
Can I ask something @philippmeisberger ? Is this library works with PFM10 and R305 fingerprint module?

philippmeisberger, 2017-03-02:
I checked the specifications and yes the R305 fingerprint module should work but the PFM10 will not

rubiagatra, 2017-03-06:
Thank you @philippmeisberger it is works on R305. Can I ask some question? What is PyFingerprint database?, I am relative new about this kind of stuff. I already read the code, it mentions about charbuffer but, I still dont know about it. Could you tell me the resources that I need to understand the code about where is the fingerprint stored or saved.

Thank You

closed 2017-03-16 by bastianraschke

EsEm17, 2017-03-24:
We have encountered this error:
Operation failed!
Exception message: storeTemplate() takes at least 2 arguments (1given)
Can you help us out? Pleease. Thank you.