bastianraschke/pyfingerprint #9: "fingerprint sensor could not be initialized! Exception message: the fingerprint sensor port "dev/ttyUSB0" was not found"

tatedobs, 2017-03-09:
I am trying to set up the fingerprint sensor on a Raspberry Pi 3 and am getting this error when I try running any example file. I have tried changing the port to dev/ttyAMA0 but when I do this the python file never actually runs and I never get any feedback. I am kind of unsure about what this port actually means and which pins on the Raspberry Pi I should be plugging in to.

bastianraschke, 2017-03-09:
Please checkout the section "Update 7" on my blog post https://sicherheitskritisch.de/2014/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux/ for instructions to get the sensor working with the Raspberry Pi 2/3.

closed 2017-03-16 by bastianraschke

yanickrochon, 2019-12-13:
This does not help; there is no `/dev/ttyUSB0` after everything.