bastianraschke/pyfingerprint #17: Importer cannot find module named utilities

ijm329, 2017-04-15:
Hello, I just cloned this project today and followed all of the installation instructions and whatnot. However, when I try to run any of the examples I get the following error: 

`Traceback (most recent call last):
  File "/usr/share/doc/python-fingerprint/examples/example_enroll.py", line 14, in <module>
    from pyfingerprint.pyfingerprint import PyFingerprint
  File "build/bdist.linux-armv6l/egg/pyfingerprint/pyfingerprint.py", line 14 , in <module>
ImportError: No module named utilities`

How do I go about fixing this? I've attempted to add the working directory to the python path, but that doesn't seem to work either.

philippmeisberger, 2017-04-16:
Hello @ijm329. Which Python version are you using? It seems that you are using Python3. PyFingerprint 1.4 only supports Python2. We are currently working on the next release which will support Python3.

bastianraschke, 2017-04-16:
Please test with newest Development @ijm329

bastianraschke, 2017-04-23:
Should be fixed. I close the issue cause of any reaction.

closed 2017-04-23 by bastianraschke