bastianraschke/pyfingerprint #19: After using setAddress() sensor will ignore any command

philippmeisberger, 2017-04-16:
Internal cached address is not updated so old address is used in following commands. This provokes the sensor to ignore commands.

bastianraschke, 2017-04-16:
Thanks. Can we fix it when we update the `__address` field?

philippmeisberger, 2017-04-20:
It is a possible bug, not an actual issue. I tried the following on my ZFM-20:

```
from pyfingerprint.pyfingerprint import PyFingerprint

f = PyFingerprint('/dev/ttyUSB0', 57600, 0xFFFFFFFF, 0x00000000)

if ( f.verifyPassword() == False ):
    raise ValueError('The given fingerprint sensor password is wrong!')

f.setAddress(0xFFFFFFFE)
f.readImage()

## Reset address to default
f.setAddress(0xFFFFFFFF)
```

Sensor still accepts the old address and `readImage()` is executed like expected (light turns on). But other models of ZhianTec (e.g. ZFM-60) may behave different. For convience: Update the `__address` field.

commit f89e8cbe (2017-04-22): Fixed #19 - After using setAddress() sensor will ignore any command
closed 2017-04-22 by bastianraschke