bastianraschke/pyfingerprint #20: After using setPassword() sensor will ignore any command

philippmeisberger, 2017-04-16:
Internal cached password is not updated so old password is used in following commands. This provokes the sensor to ignore commands.

bastianraschke, 2017-04-16:
Thanks. Can we fix it when we update the `__password` field and manual call `verifyPassword()`?

philippmeisberger, 2017-04-17:
Yes, but calling `verifyPassword()` is not necessary

philippmeisberger, 2017-04-17:
Shall I push change on Development branch?

philippmeisberger, 2017-04-20:
It is a possible bug, not an actual issue. I tried the following on my ZFM-20:

```
from pyfingerprint.pyfingerprint import PyFingerprint

f = PyFingerprint('/dev/ttyUSB0', 57600, 0xFFFFFFFF, 0x00000000)

if ( f.verifyPassword() == False ):
    raise ValueError('The given fingerprint sensor password is wrong!')

f.setPassword(0x00000001)
f.readImage()

## Reset password to default
f.setPassword(0x00000000)
```

Sensor still accepts the old password and `readImage()` is executed like expected (light turns on). But other models of ZhianTec (e.g. ZFM-60) may behave different. For convience: Update the `__password` field.

commit 6e262ca9 (2017-04-22): Fixed #20 - After using setPassword() sensor will ignore any command
closed 2017-04-22 by bastianraschke