bastianraschke/pyfingerprint #26: Unknown error 0x17

wsslfnstr, 2018-01-06:
Im getting this error when i loop it some times it gives this error the first time and sometime after 3 times etc. How to fix this?

Traceback (most recent call last):
  File "School.py", line 88, in <module>
    startApp()
  File "School.py", line 68, in startApp
    result = f.searchTemplate()
  File "/usr/lib/python2.7/dist-packages/pyfingerprint/pyfingerprint.py", line 956, in searchTemplate
    raise Exception('Unknown error '+ hex(receivedPacketPayload[0]))
Exception: Unknown error 0x17

philippmeisberger, 2018-01-06:
Which sensor are you using?

wsslfnstr, 2018-01-06:
Im using this one: FPM10A Fingerprint Reader Sensor Module
Its working bus sometimes crashes with the error code!

P.S. compliments for the fast replay

https://www.aliexpress.com/item/FPM10A-Fingerprint-Reader-Sensor-Module-Optical-Fingerprint-Fingerprint-Module-For-Arduino-Locks-Serial-Communication-Interface/32834946780.html?spm=a2g0s.9042311.0.0.tMZROE

philippmeisberger, 2018-01-06:
First I want to say I have no idea what is going wrong but I try to help you. The error seems not to be a ZhianTec error code as is not defined. To find out what the error code 0x17 stands for I tried to find a data sheet for the FPM10A and found this http://www.gemmarduino.net/PM1/00100-145/1307061015.pdf. Poorly the error code is not specified there. Maybe you have a datasheet? Please remind that the library is designed for ZhianTec fingerprint modules.

wsslfnstr, 2018-01-06:
Im sorry i dont have a datasheet

bastianraschke, 2018-01-31:
I guess the sensor is not compatible, thus i close this issue.

closed 2018-01-31 by bastianraschke

vxander, 2020-02-19:
17H：表示残留指纹或两次采集之间手指没有移动过
https://wenku.baidu.com/view/c5c2b2f1da38376baf1faee5.html
Indicates that the finger has not moved between the two acquisitions