bastianraschke/pyfingerprint #33: Implementation of uploadImage()

torehc, 2018-02-19:
Hi! First congratulations for the work.

I would like to know if you plan to implement the uploadImage() function.
I think it's very interesting to have 2 separate fingerprint readers who can share information.

Greetings.

philippmeisberger, 2018-02-19:
Hi @torehc . `uploadImage()` may be implemented in a future release but currently there is no need for it. As the method is too specially the priority is low. For uploading templates you could use `uploadCharacteristics()` instead.

torehc, 2018-02-19:
it's fine. Thanks for the quick response.

moiseslodeiro, 2018-03-19:
Same here requesting that function... Is this too difficult to implement?

philippmeisberger, 2018-03-19:
I already implemented `uploadImage` for my Delphi based ZFM library and this was no fun: https://github.com/philippmeisberger/zfmsdk/blob/b154a730970a0de4ce9bca2ca37e0516ed2ca883/src/PMCW.Serial.ZFM.pas#L1772
I still want to port the code to Python but did not find time to do this yet. PR is appreciated.

As I already mentioned: this function is so specially. Why not using `uploadCharacteristics` instead?

bastianraschke, 2018-06-19:
Because of no responses, i close the issue now.

closed 2018-06-19 by bastianraschke