bastianraschke/pyfingerprint #34: The sample scripts work only when you run them three times

felipechp, 2018-03-02:
Hi bastianraschke,

I have the FPM10A Fingerprint Reader Sensor Module. When I run the password script the first time, the script waits for an answer after sending the password and it does not continue. The second execution of the script does not work either and I get the answer:

"The fingerprint sensor could not be initialized!
Exception message: The received packet do not begin with a valid header!"

The third execution does work correctly as expected and the sensor light turns on. The following executions of the script work correctly.

This behavior is repeated when I turn off the sensor and turn it on again by removing the 5v power supply from the sensor. This always happens with the AMA0 serial port of the Raspberry and also with the USB0 port with a CH340 serial converter. This behavior is the same with other computers with Ubuntu and with the execution of the other example scripts.

Could you give me a clue how to solve this problem?

Thank you very much and congratulations for the work.

DavidAshraf, 2018-04-06:
Hi felipechp
Have you found a solution to this yet ?

bastianraschke, 2018-06-19:
Sorry i just have the tips listed here: https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/ FAQ #1

closed 2018-06-19 by bastianraschke