bastianraschke/pyfingerprint #36: Compare BMP with fingerprint

wsslfnstr, 2018-04-11:
Hello,
I want to compare a downloaded fingerprint (.BMP) with a fingerprint that is scanned.
Is this posible?

 Im using a raspberry pi 3 and code:

`import time
from pyfingerprint.pyfingerprint import PyFingerprint
try:
    f = PyFingerprint('/dev/ttyUSB0', 57600, 0xFFFFFFFF, 0x00000000)

    if ( f.verifyPassword() == False ):
        raise ValueError('The given fingerprint sensor password is wrong!')

except Exception as e:
    print('The fingerprint sensor could not be initialized!')
    print('Exception message: ' + str(e))
    exit(1)
try:
    print('Waiting for finger...')
    while ( f.readImage() == False ):
        pass

    f.convertImage(0x01)

    print('Remove finger...')
    time.sleep(2)

    print('Waiting for same finger again...')
    while ( f.readImage() == False ):
        pass

    f.convertImage(0x02)

    if ( f.compareCharacteristics() == 0 ):
        raise Exception('Fingers do not match')

    print('match')

except Exception as e:
    print('Operation failed!')
    print('Exception message: ' + str(e))
    exit(1)
`
I want finger 2 to be a file instead of a finger.

philippmeisberger, 2018-04-15:
Hello,
this was already asked in #33 

Why not using sensor database? Why not using characteristics instead of images?

bastianraschke, 2018-06-19:
Because of no responses, i close the issue now.

closed 2018-06-19 by bastianraschke