bastianraschke/pyfingerprint #39: May I use this library with GT511C3 by Sparkfun.

VIkash2601, 2018-05-23:
Sir I'm using GT511C3 with Raspberry Pi3. I'm looking for the same functioning as you have mentioned in your library. So do I use the same codes or should I change something before using them?

Thanks in advance.

philippmeisberger, 2018-05-23:
Hi @VIkash2601 ,
regarding the datasheet (https://cdn.sparkfun.com/assets/learn_tutorials/7/2/3/GT-521F52_Programming_guide_V10_20161001.pdf) the GT511C3 is not compatible with PyFingerprint. There seems to be a separate SDK (https://cdn.sparkfun.com/assets/learn_tutorials/7/2/3/20171129-SDK_Demo_Ver1.9.zip).

VIkash2601, 2018-06-15:
So what should I do to use this project? Is there any library.
With Regards
Vikash Sharma
88299028689929172055

philippmeisberger, 2018-06-15:
As I already said, your sensor is not compatible with this library so you cannot use this project. There is an other library you could use: https://cdn.sparkfun.com/assets/learn_tutorials/7/2/3/20171129-SDK_Demo_Ver1.9.zip

closed 2018-06-19 by bastianraschke