bastianraschke/pyfingerprint #40: How to store fingerprint templates in database

prabhakaran4920, 2018-05-31:
Hi,
R305 is only capable for storing 256 templates, I want to store more templates . So i planed to store fingerprint  templates in database. But I don't know how to do that... can anyone help me do that..

philippmeisberger, 2018-05-31:
Take a look at `example_enroll.py`: Let the sensor create the template for you https://github.com/bastianraschke/pyfingerprint/blob/0432a92251c1b88fae3e265a07c5c5c9cf76b438/src/files/examples/example_enroll.py#L69 but instead of storing it with `storeTemplate()` https://github.com/bastianraschke/pyfingerprint/blob/0432a92251c1b88fae3e265a07c5c5c9cf76b438/src/files/examples/example_enroll.py#L72 use `downloadCharacteristics()` to download and store it locally. For comparing two templates you must first upload them using `uploadCharacteristics()` and use `compareCharacteristics()`. Could possibly work.

chimenk, 2018-06-12:
Hello,

I'm saving the downloaded characteristics into MySQL text column. When i try to upload those characteristics like this :

`f.uploadCharacteristics(0x02, chr)`

The chr variable is returned value from MySQL which contains the characteristics, this give me error "cannot convert argument to integer".  And when i cast the chr variable to integer then this error is shown "invalid literal for int() with base 10".

Any suggestion? i think i'm doing something wrong here.

bastianraschke, 2018-06-19:
So it seems resolved - i close it.

closed 2018-06-19 by bastianraschke

kanybeka13, 2020-01-29:
> Hello,
> 
> I'm saving the downloaded characteristics into MySQL text column. When i try to upload those characteristics like this :
> 
> `f.uploadCharacteristics(0x02, chr)`
> 
> The chr variable is returned value from MySQL which contains the characteristics, this give me error "cannot convert argument to integer". And when i cast the chr variable to integer then this error is shown "invalid literal for int() with base 10".
> 
> Any suggestion? i think i'm doing something wrong here.
> 
> Updated:
> After reading this [https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/](url) i can upload and finally compare the characteristics. Only need to add eval function. Thanks

Link not a work anymore. Can you describe in more detail what you did to upload templates to the scanner, or can you provide your code?

adityadate1997, 2020-03-13:
I want to download and store the template in my server datatabase, I am using NodeMCU and Php with mysql on server side. This function is not provided in the library. Can anybody please guide me on this?

philippmeisberger, 2020-03-14:
https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/

nguyenhaibien, 2020-06-11:
> https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/

hello! 
I'm using RPi3 and R308 sensor to execute example_search.py but it alway return Exception error: Communicate error
and fail at this line:
 characterics = str(f.downloadCharacteristics(0x01)).encode('utf-8')
please help me to fix it? thank you

philippmeisberger, 2020-06-12:
> hello!
> I'm using RPi3 and R308 sensor to execute example_search.py but it alway return Exception error: Communicate error
> and fail at this line:
> characterics = [...]

Please see #44

nguyenhaibien, 2020-06-17:
Hi!
Many thanks for replying to my comment, I will follow your instructions
(Sorry for the late reply and my English is everage)

inayatku, 2021-11-14:
> Hello,
>
> I'm saving the downloaded characteristics into MySQL text column. When i try to upload those characteristics like this :
>
> `f.uploadCharacteristics(0x02, chr)`
>
> The chr variable is returned [...]

I think this issue is resolved . But the information  is a bit scattered. Plz let me know the actual instructions to do so. If possible plz share a sample python code to upload and download fingerprint characteristic from mysql database.  Your help would be much appreciated.