bastianraschke/pyfingerprint #41: Sensor not responding after setSystemParameter()

daniel-blazquez, 2018-06-19:
Hi,
library was working great (thanks) but a bit slow for download image (7-8 seconds). I tried using setSystemParameter(4, 12) to increase the baud rate, and since then, the sensor is not longer responding. Any ideas? any way to reset the sensor to factory settings? 
thanks Bastian!
Daniel

bastianraschke, 2018-06-25:
Hey Daniel, i don't know an option to reset the sensor to factory settings. I never increased the sensors baudrate because i didn't need it.

Did you get an OK from the sensor after you did the "setSystemParameter" call? You power disconnected the sensor after you set the parameter? And you tried to connect to the sensor with exactly 115200 Bauds?

daniel-blazquez, 2018-06-25:
Thanks for getting back to me, Bastian. The sensor is working fine now at 115200 thru pyfingerprint. Regardless, I spent some time looking for ways to erase the flash of the FZ1035G and found nothing. I will post back if I do find anything! Cheers!

closed 2018-06-25 by daniel-blazquez