bastianraschke/pyfingerprint #42: Same fingerprint on different fingerprint sensors

MajidAlaeinia, 2018-07-11:
I am using R303 fingerprint sensor and it works properly with the library. **I have a database table for the fingerprints, and each fingerprint is stored in the database with the related fields.**

I want to use another R303 sensor, and not to `enroll` the old finger into the new sensor. I mean to enroll a fingerprint in one sensor and be able to find the exact fingerprint by the new sensor.

I have read all the issues already, and seems there's no upload function. Is there any method or property in the library, that can help me with this?

philippmeisberger, 2018-07-15:
Download characteristics from sensor 1 using `downloadCharacteristics()` and upload them to sensor 2 using `uploadCharacteristics()`. After that store the uploaded template using `storeTemplate()` on sensor 2. Not tested but could possibly work.

MajidAlaeinia, 2018-07-18:
I will share the results, as soon as I buy the new sensor and test those methods.

closed 2018-07-30 by MajidAlaeinia