bastianraschke/pyfingerprint #43: Unable to communicate using Raspberry Pi UART

bose0003, 2018-07-26:
Hello, 
I have recently started working on R307 fingerprint module and have been trying to connect it to my Raspberry Pi 3 through the UART (not USB). I am not using any TTL for now.
I am able to connect the Pi and the sensor using UART but am not able to have may meaningful communication. Every time I send any command to execute, I get: 

```
Traceback (most recent call last):
 File "main.py", line 14, in <module>
   obj.clearDatabase()
 File "/home/pi/Biometry_Attendance_System/Attendance/pyattendance1.py", line 1101, in clearDatabase
   raise Exception('Unknown error '+ hex(receivedPacketPayload[0]))
Exception: Unknown error 0x21
```

I do not know what might be the issue and would really appreciate any help that I can get. 
Hope I could find some help here.
Thanks.

philippmeisberger, 2018-07-28:
Please ensure you have connected the sensor correctly and checkout the section "Update 7" at https://sicherheitskritisch.de/2014/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux/ for instructions to get the sensor working with the Raspberry Pi 3.

bose0003, 2018-07-28:
Thanks mate. I was finally able to work it out and the sensor is now communicating with the Pi successfully. I am now trying to resolve a "invalid header" error, but we can close this as resolved. Thanks again.

closed 2018-07-28 by bose0003