bastianraschke/pyfingerprint #44: R308 sensor not working properly (Exception message: Communication error)

roman-podorvanov, 2018-09-12:
Every given example fails at this line:
```
print('Currently used templates: ' + str(f.getTemplateCount()) +'/'+ str(f.getStorageCapacity()))
```
getStorageCapacity calls self.getSystemParameters()[2] which raises exception at line 558

Everything else I've tested (f.getTemplateIndex(page), f.readImage(), f.createTemplate(), f.compareCharacteristics()) works fine. So I think the only problem is getSystemParameters method.

My fingerprint sensor: 
https://aliexpress.com/item/R308-Biometric-Fingerprint-Module-Sensor-reader-Scanner/32605490463.html

Manuals: https://www.dropbox.com/sh/orprmb3bgb6lqb6/AABJrv83ZyzCKCYFGlcjigi0a/User%20Manual%20for%20Fingerprint%20Module?dl=0&subfolder_nav_tracking=1

SDKs some chinese guy gave me:
https://www.dropbox.com/sh/pznvlzx8qx5nfr3/AABpzhSyjqH0qWNYgMvxqAA9a?dl=0

philippmeisberger, 2018-09-12:
I checked the datasheet: The R308 does not support FINGERPRINT_GETSYSTEMPARAMETERS message. That is the reason `getSystemParameters()` does not work.

roman-podorvanov, 2018-09-12:
Yeah I checked datasheets too and did not find any SYSTEMPARAMETERS or 0x0F related stuff for either of these sensors (r303, r305 etc) however they works just fine. I'm sure there is some way to get those parameters for r308 too but I don't have any experience neither with python nor any other hardware related stuff so I don't know what to look for. Maybe you can suggest me to try something?

philippmeisberger, 2018-09-12:
Look at the last two pages in every datasheet ;) There is a list of all supported instructions. `ReadSysPara` (instruction 0x0F) is supported by R301T, R302, R303, R305, R306, R307 and R551. Poorly R308 seems not to support this instruction.

roman-podorvanov, 2018-09-12:
You're right. I think I don't really need getSystemParameters method anyway. I've just hardcoded some values inside pyfingerprint.py:

 *line 1209 (getMaxPacketSize)*
```
##packetMaxSizeType = self.getSystemParameters()[5]
packetMaxSizeType = 3
```

*line 1228 (getStorageCapacity)*
```
##return self.getSystemParameters()[2]
return 1000
```

Now I can succesfully run enroll, search and delete examples. Gues we can close this issue. Maybe you should make some notes in wiki pages.

Thank you for a great library!

closed 2018-11-12 by bastianraschke

SaJJaDAgEnT47, 2019-12-30:
> getStorageCapacity

i did this work but still getting the same error (The fingerprint sensor could not be initialized!
Exception message: The received packet do not begin with a valid header!)

please help