bastianraschke/pyfingerprint #46: R307 The fingerprint sensor could not be initialized! Exception message: The received packet do not begin with a valid header!

JuggernautPanda, 2018-10-08:
The fingerprint sensor could not be initialized!
Exception message: The received packet do not begin with a valid header!

Raspbian scratch 
Rasp pi 3 b+ and b
Exchanging the pins also does not work

philippmeisberger, 2018-10-08:
Please take a look at #1

JuggernautPanda, 2018-10-11:
Hello.  I have found that the hardware had an issue with the jumper cables that were being used. Good quality jumper cables have solved the issue.

closed 2018-11-12 by bastianraschke

sand007man, 2018-12-24:
Hi i am getting the same error when i run the exampledowImage but in Raspberry pi zero !! the same code was working properly in Raspberry pi 3 ....is it a Hardware issue or Raspberry pi zero cant handle it because i am able to enroll it but downloadImage not working?

philippmeisberger, 2018-12-24:
You already opened #52 for this

cjoshi015, 2019-05-06:
I have also encountered this problems

use this pin numbering 

1 | 5V | Regulated 5V DC
2 | GND | Common Ground
3 | TXD | Data output - Connect to MCU RX
4 | RXD | Data Input - Connect to MCU TX
5 | TOUCH | Active Low output when there is touch on sensor by finger
6 | 3.3V | Use this wire to give 3.3V to sensor instead of 5V

and also check your jumber cables are connected properly or not majorly it is due to loose connection and on many sited pin number is given wrong use above pin numbers

I have solved my issue using above steps