bastianraschke/pyfingerprint #50: downloadImage() does not work correctly with R305

TehseenHasan, 2018-11-14:
From this Issue #48 we figured out that **`downloadImage()`** function is giving wrong and c**orrupted image** of scanned finger for R305 Fingerprint Sensor... Other sensor like R301T is giving correct fingerprint image successfully without errors or data corruption... This is a sample image generated by **R305** sensor using  `example_downloadimage.py` example script:

[image]

While same example code is giving **correct image on R301T sensor**... Please solve this issue ... the error seems to be in somewhere in **`def downloadImage(self, imageDestination)`** function in **`pyfingerprint.py`** library file where it downloads image data from sensor image buffer. Thanks...

philippmeisberger, 2018-11-14:
It seems that R305 sends an image buffer that differs from ZFM models. I guess that the image resolution should be calculated dynamically. I tried to implement this in a new `downloadImage()` function. It still works on my ZFM-20. Since I got no R305 could you please try it and report if it works:

```
def downloadImage(self, imageDestination):
        """
        Download the image of a finger to host computer.

        @param string imageDestination
        @return void
        """

        destinationDirectory = os.path.dirname(imageDestination)

        if ( os.access(destinationDirectory, os.W_OK) == False ):
            raise ValueError('The given destination directory "' + destinationDirectory + '" is not writable!')

        packetPayload = (
            FINGERPRINT_DOWNLOADIMAGE,
        )

        self.__writePacket(FINGERPRINT_COMMANDPACKET, packetPayload)

        ## Get first reply packet
        receivedPacket = self.__readPacket()

        receivedPacketType = receivedPacket[0]
        receivedPacketPayload = receivedPacket[1]

        if ( receivedPacketType != FINGERPRINT_ACKPACKET ):
            raise Exception('The received packet is no ack packet!')

        ## DEBUG: The sensor will sent follow-up packets
        if ( receivedPacketPayload[0] == FINGERPRINT_OK ):
            pass

        elif ( receivedPacketPayload[0] == FINGERPRINT_ERROR_COMMUNICATION ):
            raise Exception('Communication error')

        elif ( receivedPacketPayload[0] == FINGERPRINT_ERROR_DOWNLOADIMAGE ):
            raise Exception('Could not download image')

        else:
            raise Exception('Unknown error '+ hex(receivedPacketPayload[0]))

        imageData = []

        ## Get follow-up data packets until the last data packet is received
        while ( receivedPacketType != FINGERPRINT_ENDDATAPACKET ):
            receivedPacket = self.__readPacket()

            receivedPacketType = receivedPacket[0]
            receivedPacketPayload = receivedPacket[1]

            if ( receivedPacketType != FINGERPRINT_DATAPACKET and receivedPacketType != FINGERPRINT_ENDDATAPACKET ):
                raise Exception('The received packet is no data packet!')

            imageData.append(receivedPacketPayload)

        ## Initialize image
        resultImage = Image.new('L', (len(imageData[0]) * 2, len(imageData)), 'white')
        pixels = resultImage.load()

        for y in range(len(imageData)):
            line = 0

            for x in range(0, len(imageData[y]) * 2, 2):

                ## On byte contains two pixels
                ## Thanks to Danylo Esterman <soundcracker@gmail.com> for the "multiple with 17" improvement:
                pixels[x, y] = (imageData[y][line] >> 4) * 17
                pixels[x + 1, y] = (imageData[y][line] & 0x0F) * 17
                line += 1

        resultImage.save(imageDestination)
```

Just replace the existing `downloadImage()` in your `pyfingerprint.py`.

TehseenHasan, 2018-11-14:
I tried this new code on R305...  This time there is no `image index out of range` error... It generates corrupted fingerprint image as before....but change is in the **resolution** of generated image. This time this code make image of resolution : **64x1152 pixels** ... This is the image generated using R305 sensor:
[image]

While R301T is working same as before on this new code... For R301T it is giving correct image...having resolution same as before `(256, 288) pixels`...

philippmeisberger, 2018-11-14:
Ok, so the sensor seems to provide images in resolution **1024x1152** instead of 256x288 like ZFM modules do it... 

Could you try the [SFG Demo](http://www.adafruit.com/datasheets/SFGDemoV2.0.rar) if downloading image works with this?

TehseenHasan, 2018-11-14:
Yes SFG Demo successfully capturing and saving image... 
[image]

TehseenHasan, 2018-11-22:
Wait a sec....!!!  Today I just tried again ... R305 sensor **now working** on both Windows and Raspberry Pi... image downloading **without** `image index out of range` error... How is this possible that **same sensor and same code** was not working before but today it is working perfectly... !!! 🤔

philippmeisberger, 2018-11-23:
Sounds great! But strange... Did you revert all your local changes?

TehseenHasan, 2018-11-24:
No I did not change anything... I just tried after few days... I don't know how but now It is working...