bastianraschke/pyfingerprint #52: "The received packet do not begin with a valid header!" using Raspberry pi zero w with R307 fingerprint Sensor

sand007man, 2018-12-17:
The same code works properly with Raspberry pi 3 but its throws invalid header exception with raspberry pi zero ....is there any changes i need to make ??
I already made required changes in
1)cmdline.txt
2)config.txt
and Serial0 is using TTYAMAO Really need help here

philippmeisberger, 2018-12-24:
I am sorry, but we have no Raspberry Pi Zero for investigation...

sand007man, 2018-12-26:
can this error help you give a hint where exactly the problem is 
"Currently used templates: 3/1000
Waiting for finger...
Downloading image (this take a while)...
Operation failed!
Exception message: The received packet is corrupted (the checksum is wrong)!"

philippmeisberger, 2018-12-26:
This looks good in general. The connection seems to be working ;) But the corrupted packet exception is really strange...

sand007man, 2018-12-26:
do you think my Serial is not working properly because of that the packets I am receiving is not complete or I am missing a few packets ...that's why this condition 

 if ( receivedChecksum != packetChecksum ):
                    raise Exception('The received packet is corrupted (the checksum is wrong)!')
is getting satisfied?? 
I am using a micro -usb to USB convertor to connect my fingerprint.

philippmeisberger, 2018-12-27:
I do not know... Hard to guess.

You are not using an **USB-TTL** converter right? Is downloadImage() the only method which does not work? Does downloadCharacteristics() work?

TehseenHasan, 2018-12-28:
I also got this error message when I set my R305 fingerprint sensor baudrate to 960000 bps... Then I changed baudrate to again 115200 or 57600 bps then that error gone.... then I set baudrate again to 960000 bps and changed my of USB to Serial converter (Ch340) then again this error gone...

TehseenHasan, 2018-12-28:
I also encountered this **Exception message: The received packet is corrupted (the checksum is wrong)!"** error... so maybe there is some logical issue in the code or maybe some communication related amendments in the code are required...

sand007man, 2018-12-29:
@TehseenHasan Hi my issue got sorted when I changed my Raspberry pi zero w may be the one I was running code on had some hardware defect.But my fingerprint doesnt work on any other baud rate apart from 57600!! Strange !!!

TehseenHasan, 2018-12-29:
R307 should support standard baudrates in the form of 9600*N where N<=12 as described in datasheet. I think its your Serial interface of Raspberry pi that is limiting the baudrate... try using USB to Serial Converter it should work of many baudrates. BTW baudrate is not a critical point if you are using your sensor to  just get matched index of a person and other normal opertions like enrolling, deletion. Baudrate only matters if you want to download scanned fingerprint image into Raspberry pi or PC. Then more baudrate means faster image download.

philippmeisberger, 2018-12-29:
BTW: The maximum supported baudrate of the sensor is 9600*12=115200. Anything beyond this will not work ;)

sand007man, 2019-01-01:
@TehseenHasan Hi i am using Tx and Rx pin directly  for communication between fingerprint sensor and Raspberry pi zero w ....i am not using any Serial -to ttl Convertor in between yet i am not able to work with any other baudrate apart from 57600 ...if i change the baud rate then it throws packet exceptions

sand007man, 2019-01-05:
@TehseenHasan @philippmeisberger hi if I want to change the baud rate at which the fingerprint Download the images How to change it ?? I have already tried " stty -F /dev/ttyAMA0 115200 " and after that if i change the baud rate in code ...it again throws "Invalid header " error ....and it works fine with 57600 baud rate ...i am using raspberry pi zero w .

philippmeisberger, 2019-01-05:
You will need to change the sensor baudrate using `setSystemParameter(4, N)` where N is in between 1 and 12. After that use the new baudrate (9600 * N) in your code.

abdullahsurati, 2019-01-05:
Offtopic question - How to connect R307 directly to the Pi without using the USB to TTL Convertor? Do I have to short the two pins on the R307 that says 3.3V? I am refferring to this - https://www.sunrom.com/media/content/1720/pinouts.jpg
Currently I am using Pins 1, 2, 3 and 4 of the R307 to work with the USB-TTL but I want to eliminate it.

sand007man, 2019-01-07:
@philippmeisberger Thank you its working great but is there someway i can speed up the downloadImage process ...with baud rate set to 115200 it takes 20 seconds to download the image ...i am using raspberry pi zero w!!

sand007man, 2019-01-07:
@abdullahsurati If you short that pin as per the document then pin one will become your 3.3v supply and rest will remain the same ..but if you are using 5v then you have to use voltage divider on Tx pin to 3.3v and connect fingerprint to raspberry pi.

To connect fingerprint directly to Serial port of raspberry pi follow this step:
Step-1  sudo nano /boot/config.txt
 
enable_uart=1

sudo reboot

Step-2 sudo systemctl stop serial-getty@ttyS0.service

Step-3 sudo systemctl disable serial-getty@ttyS0.service

Step-4 sudo nano /boot/cmdline.txt

Remove : console=serial0,115200

Step-5 sudo nano /boot/config.txt

dtoverlay=pi3-miniuart-bt

sudo reboot

sand007man, 2019-01-07:
and then in code change "dev/ttyUSB0" to "dev/ttyAMA0"

philippmeisberger, 2019-01-07:
@sand007man There is no way I know to speed up `downloadImage()`. But there is `downloadCharacteristics()` which is really fast. Why not using this?

TehseenHasan, 2019-01-07:
@philippmeisberger @abdullahsurati @sand007man 
Image Download speed is what I was also concerned about... It takes a lot of time to transfer scanned fingerprint image from sensor to PC or Raspberry pi. Image has about 70 kb size while baudrate is 115200 bits per second. That's why image transfer process is too slow. 
There are two solutions to increase the download speed of image: 
### **Solution 1: Increase the baud-rate:**
I used this approach in my last project where I wanted faster image download speed, I read the documentation of SynoChip **AS601** Chip used in most R30x series fingerprint sensors e.g. R305, R307 or R301T etc. Its documentation says that this chip can operates at maximum 1 Mbaud (Mega baud) so I thought to increase the baudrate more than 115200. I changed the baudrate of sensor to **960000 bps** (9600 * 100) by setting N=100 in `setSystemParameter(4, N)` function.
I also changed the code of **`setSystemParameter`** function in **`pyfingerprint.py`** file to accept N larger than 12. So change this line: 
`if ( parameterValue < 1 or parameterValue > 12 ):` 
to 
`if ( parameterValue < 1 or parameterValue > 101 ):`

Now you can pass N up to 100. The speed of image transfer is now increased a lot. And now it takes about **2 seconds only** to save fingerprint bitmap image to storage at N=100 (960000 bps baudrate).

### **Solution 2: Use USB protocol instead of UART**
**Note:** I did not completely implemented this method until now but **I'm currently working on its code**. 
I found that the USB protocol can be used by using USB pins ( 5V, D+, D-, GND) on sensor's PCB. Use an ordinary USB cable, cut one end of cable and solder the 4 wires on Sensor's PCB on these USB pins and plug other end to directly on USB port on PC or Raspberry pi.

[image]

 If you solder pins correctly you will see that the sensor will shown in ThisPC as CD-ROM drive as shown in picture below:

[image]

 If it do not show in ThisPC or in Deceive Manager or you are using OS other than Windows 10 maybe you have to install **USB drivers** of sensor (find drivers in `TestDemo_V3.012.zip\TestDemo_V3.012\driver` folder in SDK files). 
Next step is to control the sensor. I found some SDKs to control the sensor. For testing purpose check your sensor using this Software **TestDemo.exe** (from **TestDemo_V3.012.zip**) to ensure that USB connection is working correctly. Plug sensor's USB to PC open TestDemo.exe click on Open Device button and your sensor will be connected now try to capture fingerprint image by pressing Capture button, you will see that image is transferred in less than a second. 🙂

[image]

**Work to do...!**
Now last steps are to understand the SDK code for Windows or Linux so that you can integrate the code with your application. I'm currently trying to understand the working of these SDKs and I'm not a good developer if someone wants to help and work to implement this SDK into some proper form so that anyone can use them as simple as pyFingerprint Library then it will be very helpful.

### **Download the SDKs from this link:**
https://www.dropbox.com/sh/pznvlzx8qx5nfr3/AABpzhSyjqH0qWNYgMvxqAA9a?dl=0
Or from here (Alternate Link):  https://www.dropbox.com/sh/2o8to0omv2dx8ph/AAC_w_Qpw4OK42O-SStlHnZsa?dl=0

There is **another C# based code** too, download and understand it: https://www.dropbox.com/s/jc4mx7xmju6wieb/Fingerprint%20C%23%20example.zip?dl=0

sand007man, 2019-01-07:
@TehseenHasan hi i followed your method and change the value from 12 to 100 and again rebuild the packages and in the code also i change the set parameter to 100 and then baud rate to 960000 but i was getting this exception
"The fingerprint sensor could not be initialized!
Exception message: The given baudrate is invalid!
"
and when i tired to change baud rate back to 115200 its not working anymore now.. can you please tell me proper step to change the baudrate?? ...i am using r307 with synochip AS606!!

philippmeisberger, 2019-01-07:
@TehseenHasan Thank you providing your lessons learned. Can this library be used for direct USB connection? Why using another library? BTW: I also developed an SDK: https://github.com/philippmeisberger/zfmsdk

sand007man, 2019-01-07:
@philippmeisberger is  downloadCharacteristics() is same like template ?? because in database i am storing images and comparing both images. so i am kind of forced to use image !!

TehseenHasan, 2019-01-07:
@sand007man **`downloadCharacteristics()`** is the function to download **`template`**. You can not use templates to compare them in external code. You can only compare templates **within sensor** because these templates are **Proprietary Templates** means these templates can only be encoded and decoded inside sensor firmware and only manufacturer of the sensor knows how to encode and decode these templates. 
For further understanding read this thread: https://sourceforge.net/p/sourceafis/discussion/1051112/thread/72795340be/#a0e5

TehseenHasan, 2019-01-07:
> @TehseenHasan Thank you providing your lessons learned. Can this library be used for direct USB connection? Why using another library? BTW: I also developed an SDK: https://github.com/philippmeisberger/zfmsdk

Yes these SDK can be used for both Serial (UART) as well as direct USB connection. You do not need **USB to Serial Converter** or **UART interface (Tx, Rx)** if you use direct USB connection by using (5V, D+, D-, GND) pins. 
USB to serial converter will be used only if you have to use UART (Tx, Rx) interface. 
Both interfaces/protocols have their own pros and cons. But for scanned fingerprint Image downloading the **fastest way** is to use **USB interface**.

sand007man, 2019-01-07:
@TehseenHasan and my fingerprint sensor is not working after i did the changes mentioned by you...by setting baudrate 960000 and making required changes i am gettting ""The fingerprint sensor could not be initialized!
Exception message: The given baudrate is invalid! !! and when i tried to change my baudrate back to 115200 by again changing setbaudrate parameter ...i am getting invalid header exception!!

TehseenHasan, 2019-01-07:
> @TehseenHasan and my fingerprint sensor is not working after i did the changes mentioned by you...by setting baudrate 960000 and making required changes i am gettting ""The fingerprint sensor could [...]

For testing purpose use a **USB to Serial Converter** and then test different baudrates.... and attach USB to serial converter to USB port of your Raspberry pi and use "dev/ttyUSB0" as port .... after you successfully set the working baudrate then again shift back to your previous setup (without USB to Serial converter and "dev/ttyAMA0" as port).
Share your code please... let me see...

sand007man, 2019-01-07:
[image]

sand007man, 2019-01-07:
after making changes in the src files (pyfingerprint.py) do i have to build the whole package again??

TehseenHasan, 2019-01-07:
Change these lines in your code and write like following: 
```
baudrate = 115200
port = '/dev/ttyUSB0'
f = PyFingerprint(port, baudrate, 0xFFFFFFFF, 0x00000000)

    if ( f.verifyPassword() == False ):
        raise ValueError('The given fingerprint sensor password is wrong!')
val = f.setSystemParameter(4, 100)
print val
```
To change the baudrate of your sensor change port and baudrate value to your current values. and then Run the code if you get **True** from **`print val`** that means new baudrate is set successfully if get **False** then not.
After you successfully set the baudrate then remove/comment these lines:
```
val = f.setSystemParameter(4, 100)
print val
```
You just have to call `setSystemParameter` function once to change the baudrate not to call every time you run the code. And after you get **True** from `setSystemParameter` (means new baudrate is set successfully) then you have to power off and on the sensor so that new setting working.

TehseenHasan, 2019-01-07:
> after making changes in the src files (pyfingerprint.py) do i have to build the whole package again??

I do not know much about Linux... but you should rebuild maybe it is compulsory. Just do as it will not cost you. 🙂

sand007man, 2019-01-10:
@TehseenHasan after making the changes and running the code i get this exception
Traceback (most recent call last):
  File "example_test.py", line 21, in <module>
    if ( f.verifyPassword() == False ):
  File "/usr/lib/python2.7/dist-packages/pyfingerprint/pyfingerprint.py", line 351, in verifyPassword
    receivedPacket = self.__readPacket()
  File "/usr/lib/python2.7/dist-packages/pyfingerprint/pyfingerprint.py", line 300, in __readPacket
    raise Exception('The received packet do not begin with a valid header!')
Exception: The received packet do not begin with a valid header!

sand007man, 2019-01-11:
@TehseenHasan Hi I created a fresh os and repeated all the steps and now I am able to set baud rate but the difference in speed is not that much it use to take 20.48 seconds before and now after setting it to 960000 I am getting 20.40 !! am i suppose to change something else in raspberry pi side.I tried "stty -F /dev/ttyAMA0 960000" but it's showing invalid argument

TehseenHasan, 2019-01-11:
**As I said earlier**, please use a **USB to Serial Converter** (like CH340 , FT232 or CP2102 , **I used CH340**). And first of all, perform all experimentation by using this USB to Serial converter.... if that works then try to use Raspberry pi GPIOs Serial pins (ttyAMA0). But first try using ttyUSB0 with USB to Serial Converter...

sand007man, 2019-01-11:
@TehseenHasan Thank you for all the help and guidance. I am using FT232 USB to serial converter and when I connect it to my raspberry pi Zero board I was getting following error only for downloadimage code 
"Currently used templates: 1/1000
Waiting for finger...
Downloading image (this take a while)...
Operation failed!
Exception message: The received packet is corrupted (the checksum is wrong)!
"

So for time being, I tried to run the same code on raspberry pi 3 b+ with USB and it worked fine and when I try to make that baud rate changes it was able to download the image at around 4 seconds with baud rate of "153600" but when i ran the code with 163200 baud rate i was getting following error

Currently used templates: 0/1000
Waiting for finger...
Downloading image (this take a while)...
Operation failed!
Exception message: The received packet is corrupted (the checksum is wrong)!

TehseenHasan, 2019-01-11:
Yes I also noticed that **at higher baudrates** different USB to Serial converters behaves unexpected... I tried 960000 baudrate using a CH340 USB to Serial converter and it worked, but another same model (CH340) module did not work. Means that sensor is connected and initialized successfully **at any** baudrate and ### with any USB to Serial Converter module **but just Download Image function did not work** at some modules even same models... I also tried using FT232 USB to serial converter and it give same error as yours.... 
So I think it is some sort of **logical error related to communication in library**... Maybe it is related to serial **timeout** where some timeout occurred before complete data packet is arrived from sensor and hence `received packed is corrupted` error is occurred.... or some other error... Maybe @philippmeisberger would help us from here... @philippmeisberger Sir, please help us to solve that error.... Thanks !

sand007man, 2019-01-14:
@TehseenHasan this is out of topic but i did manage to speed up image download process by 3 seconds in raspberry pi zero by overclocking the raspberry pi using following code in config.txt
arm_freq=1050
gpu_freq=500
core_freq=500
sdram_freq=500
sdram_schmoo=0x02000020
over_voltage=2
sdram_over_voltage=2

philippmeisberger, 2019-01-14:
@TehseenHasan Maybe FT232 and some CH340 do not support 960000 baud since this is extremly high... Regarding the documentation the maximum baudrate of R307 is 115200. I suggest using this documented baudrate.

> So I think it is some sort of logical error related to communication in library...

IMHO this is not an issue of PyFingerprint ;)

TehseenHasan, 2019-01-14:
> @TehseenHasan Maybe FT232 and some CH340 do not support 960000 baud since this is extremly high... Regarding the documentation the maximum baudrate of R307 is 115200. I suggest using this documented [...]

FT232 USB to Serial Converter supports up-to **3 Mbaud** (**3145728 bps**) (According to FT232 documentation). And CH340 supports up to **2000000 bps** (**~2 Mbaud**). 
So USB to Serial Converter should not be a problem. BTW, its okay to use this library for about 99% typical projects. But sometimes there are advanced requirements like speed image transfer. 🙂

TehseenHasan, 2019-01-31:
This is a C# program to communicate with R30x Fingerprint sensor with USB protocol... This example downloads scanned fingerprint image from Sensor to our Computer....
https://github.com/TehseenHasan/R305_fingerprint_sensor_usb_interface

Understand the different functions of SynoAPIEx.dll file from this example if you want to use this same as pyFingerprint library...

bastianraschke, 2019-02-27:
I'll close the issue if no updates coming soon. Okay with that @philippmeisberger?

closed 2019-03-01 by bastianraschke

parthipanssr, 2019-03-21:
Hi 
I am trying to connect R307 Finger Print sensor to raspberry pi 3 Model B. vai GPIO Pin (5v, Gnd, TX, RX) using the instruction what you are given but I got the following error.please to help me to solve this issue.

SerailPort  /dev/ttyAMA0
Baud Rate 57600 

The fingerprint sensor could not be initialized!
Exception message: The received packet do not begin with a valid header!

sharad386, 2019-03-21:
@parthipanssr 
To connect fingerprint directly to Serial port of raspberry pi follow this step:
Step-1 sudo nano /boot/config.txt

enable_uart=1

sudo reboot

Step-2 sudo systemctl stop serial-getty@ttyS0.service

Step-3 sudo systemctl disable serial-getty@ttyS0.service

Step-4 sudo nano /boot/cmdline.txt

Remove : console=serial0,115200

Step-5 sudo nano /boot/config.txt

dtoverlay=pi3-miniuart-bt

sudo reboot

parthipanssr, 2019-03-22:
Thanks for your reply. I tried those steps but still i am getting same error. i using port  /dev/ttyAMA0 ,57600 baud rate. in my program is any changes in that.plz help me to solve this error.

sharad386, 2019-03-22:
@parthipanssr run this commands and check for serial0 and serial1 value where its pointing to tttyAMA0 or ttyS0
cd /dev/
ls -l

sharad386, 2019-03-22:
for this to work your serial0 should be pointing to ttyAMA0 if it is then there is something wrong with your fingerprint hardware or connecting cables.

parthipanssr, 2019-03-22:
serial0 pointing ttyAMA0  and serial1 pointing ttyS0. and my fingerprint module works when I am connected USB using TTL-USB converter

sharad386, 2019-03-22:
sudo stty -F /dev/ttyAMA0 57600 <- this is set the baudrate of the serial port try this...
secondly try dmesg | grep tty to check if  ttyAMA0 is assigned properly

parthipanssr, 2019-03-22:
i got following response. but same error i got
pi@raspberrypi:~ $ sudo stty -F /dev/ttyAMA0 57600
pi@raspberrypi:~ $ dmesg | grep tty
[    0.000489] console [tty0] enabled
[    0.277066] 3f215040.uart: ttyS0 at MMIO 0x3f215040 (irq = 59, base_baud = 50000000) is a 16550
[    0.831732] 3f201000.uart: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
[    1.696007] systemd[1]: Expecting device dev-ttyAMA0.device...
[    1.729476] systemd[1]: Starting system-serial\x2dgetty.slice.
[    1.729739] systemd[1]: Created slice system-serial\x2dgetty.slice.

parthipanssr, 2019-03-22:
My Python Code is

try:
    f = PyFingerprint('/dev/ttyAMA0', 57600, 0xFFFFFFFF, 0x00000000)

    if ( f.verifyPassword() == False ):
        raise ValueError('The given fingerprint sensor password is wrong!')

except Exception as e:
    print('The fingerprint sensor could not be initialized!')
    print('Exception message: ' + str(e))
    exit(1)

sharad386, 2019-03-22:
"Expecting device dev-ttyAMA0.device..." I guess this is because the hardware is not connected to the raspberry pi 3. I still feel like its a patch cord issue. You can do one thing is a test if Tx and Rx is working in Raspberry pi ..by writing simple serial code and use Ft232 or some other hardware to check if those pins are working fine because I had the same issue and it got sorted using the method I showed you

parthipanssr, 2019-03-22:
OK I WILL CHECK AND REPLY ..

Gnanavel76, 2022-11-25:
> @philippmeisberger @abdullahsurati @sand007man Image Download speed is what I was also concerned about... It takes a lot of time to transfer scanned fingerprint image from sensor to PC or Raspberry pi. Image has about 70 kb size while baudrate is 115200 bits per second. That's why image transfer process is too slow. There are two solutions to increase the download speed of image (**I tested these two methods on R305 and R301T sensors.**):
> 
> ### **Solution 1: Increase the baud-rate:**
> I used this approach in my last project where I wanted faster image download speed, I read the documentation of SynoChip **AS601** Chip used in most R30x series fingerprint sensors e.g. R305, R307 or R301T etc. Its documentation says that this chip can operates at maximum 1 Mbaud (Mega baud) so I thought to increase the baudrate more than 115200. I changed the baudrate of sensor to **960000 bps** (9600 * 100) by setting N=100 in `setSystemParameter(4, N)` function. I also changed the code of **`setSystemParameter`** function in **`pyfingerprint.py`** file to accept N larger than 12. So change this line: `if ( parameterValue < 1 or parameterValue > 12 ):` to `if ( parameterValue < 1 or parameterValue > 101 ):`
> 
> Now you can pass N up to 100. The speed of image transfer is now increased a lot. And now it takes about **2 seconds only** to save fingerprint bitmap image to storage at N=100 (960000 bps baudrate).
> 
> ### **Solution 2: Use USB protocol instead of UART**
> **Note:** I did not completely implemented this method until now but **I'm currently working on its code**. I found that the USB protocol can be used by using USB pins ( 5V, D+, D-, GND) on sensor's PCB. Use an ordinary USB cable, cut one end of cable and solder the 4 wires on Sensor's PCB on these USB pins and plug other end to directly on USB port on PC or Raspberry pi.
> 
> [image]
> 
> If you solder pins correctly you will see that the sensor will shown in ThisPC as CD-ROM drive as shown in picture below:
> 
> [image]
> 
> If it do not show in ThisPC or in Device Manager or you are using operating system other than Windows 10 maybe you have to install **USB drivers** of sensor (find drivers in `TestDemo_V3.012.zip\TestDemo_V3.012\driver` folder in SDK files). Next step is to control the sensor. I found some SDKs to control the sensor. For testing purpose check your sensor using this Software **TestDemo.exe** (from **TestDemo_V3.012.zip**) to ensure that USB connection is working correctly. Plug sensor's USB to PC open TestDemo.exe click on Open Device button and your sensor will be connected now try to capture fingerprint image by pressing Capture button, you will see that image is transferred in less than a second. 🙂
> 
> [image]
> 
> **Work to do...!** Now last steps are to understand the SDK code for Windows or Linux so that you can integrate the code with your application. I'm currently trying to understand the working of these SDKs and I'm not a good developer if someone wants to help and work to implement this SDK into some proper form so that anyone can use them as simple as pyFingerprint Library then it will be very helpful.
> 
> ### **Download the SDKs from this link:**
> https://www.dropbox.com/sh/pznvlzx8qx5nfr3/AABpzhSyjqH0qWNYgMvxqAA9a?dl=0 Or from here (Alternate Link): https://www.dropbox.com/sh/2o8to0omv2dx8ph/AAC_w_Qpw4OK42O-SStlHnZsa?dl=0
> 
> There is **another C# based code** too, download and understand it: https://www.dropbox.com/s/jc4mx7xmju6wieb/Fingerprint%20C%23%20example.zip?dl=0

I want to connect my R307 through USB interface. I soldered the wire as per the diagram and also installed the driver. But when i connect to my PC the driver manager says `Unknown USB Device Descriptor Request Failed`

pluzmedia, 2023-05-31:
> @philippmeisberger @abdullahsurati @sand007man Image Download speed is what I was also concerned about... It takes a lot of time to transfer scanned fingerprint image from sensor to PC or Raspberry [...]

After the following solution 1 increased baudrate to 960000 R3017 stopped working and sensor no more responding.

pluzmedia, 2023-06-03:
> can this error help you give a hint where exactly the problem is "Currently used templates: 3/1000 Waiting for finger... Downloading image (this take a while)... Operation failed! Exception message: The received packet is corrupted (the checksum is wrong)!"

Hi, Did you found any solution?.