bastianraschke/pyfingerprint #53: ZFM-60 | Exception message: The received packet do not begin with a valid header!

ardawan, 2018-12-26:
I'm using ZFM-60 and RPI 3B+. i followed all the README file but when i run the enroll command, i get error. I've tried both UART pins(BCM14, BCM15) and USB connection for the sensor.

```
// UART connection
$ python /usr/share/doc/python-fingerprint/examples/example_enroll.py
The fingerprint sensor could not be initialized!
Exception message: The fingerprint sensor port "/dev/ttyUSB0" was not found!

// USB connection
$ python /usr/share/doc/python-fingerprint/examples/example_enroll.py
The fingerprint sensor could not be initialized!
Exception message: The received packet do not begin with a valid header!
```
Why this is happening and how can i fix this?
Cannot use the sensor via UART instead of USB?

ardawan, 2018-12-26:
> Command :
> 
> Step-1 sudo nano /boot/config.txt
> 
> enable_uart=1
> 
> sudo reboot
> 
> Step-2 sudo systemctl stop [serial-getty@ttyS0.service](mailto:serial-getty@ttyS0.service)
> 
> Step-3 sudo systemctl disable [serial-getty@ttyS0.service](mailto:serial-getty@ttyS0.service)
> 
> Step-4 sudo nano /boot/cmdline.txt
> 
> Remove : console=serial0,115200
> 
> Step-5 sudo nano /boot/config.txt
> 
> dtoverlay=pi3-miniuart-bt
> 
> sudo reboot

Dose Not Work !

sand007man, 2018-12-26:
After that in your code change ttyUSB0 to ttyAMA0......or to check just run "sudo dmesg | grep tty" you will come to know which serial port is active

ardawan, 2018-12-26:
```
$ sudo dmesg | grep tty
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles quiet
[    0.000282] console [tty1] enabled
[    0.669795] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
[    0.671571] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 166, base_baud = 50000000) is a 16550
```

and when i run the example still checking the ttyUSB0
```
$ python /usr/share/doc/python-fingerprint/examples/example_enroll.py
The fingerprint sensor could not be initialized!
Exception message: The fingerprint sensor port "/dev/ttyUSB0" was not found!
```

sand007man, 2018-12-26:
Ok first try with usb ...once you connect usb run "sudo dmesg | grep tty" ...if you are not able to see USB0 means drivers are not installed for it....if thats the case then run "sudo apt install python-serial" and again check for the serial port !!

ardawan, 2018-12-26:
I've installed python-serial already on my machine.
Still i'm getting the same error
```
$ sudo dmesg | grep tty
[    0.000000] Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=800 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=/dev/mmcblk0p7 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait splash plymouth.ignore-serial-consoles quiet
[    0.000281] console [tty1] enabled
[    0.669066] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 87, base_baud = 0) is a PL011 rev2
[    0.670713] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 166, base_baud = 50000000) is a 16550
[    2.136102] systemd[1]: Created slice system-getty.slice.
[  739.332708] usb 1-1.3: pl2303 converter now attached to ttyUSB0
```
```
$ python /usr/share/doc/python-fingerprint/examples/example_enroll.py 
The fingerprint sensor could not be initialized!
Exception message: The received packet do not begin with a valid header!
```

ardawan, 2018-12-26:
> https://sicherheitskritisch.de/2014/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux/

im used the english version of this article. that's what i get after following the article!!

anubhaw2091, 2018-12-29:
Did anyone get any solution for this? Just bumping this up?

ardawan, 2018-12-31:
> Did anyone get any solution for this? Just bumping this up?
apparently no one supports this lib anymore...

philippmeisberger, 2018-12-31:
@ardawan Please carefully read https://sicherheitskritisch.de/2014/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux/

There must be something wrong with your hardware... Take a look at this line:
> [    0.669066] 3f201000.serial: ttyAMA0 at MMIO 0x3f201000 (irq = 87, **base_baud = 0**) is a PL011 rev2

My machine shows this:
> [    0.614124] serial8250: ttyS0 at I/O 0x3f8 (irq = 4, **base_baud = 115200**) is a 16550A

Please ensure you got everything connected as described in #1 

> apparently no one supports this lib anymore...

Of course we support this library but sometimes we have vacation ;)

closed 2019-01-04 by ardawan