bastianraschke/pyfingerprint #57: example_downloadimage.py doesn't work at all

ardawan, 2019-01-07:
If that means download to the computer, why should ask for fingerprint scanning again, why can't use existing fingerprint from its db?

im looking for a way to extract the fingerprint data to my RPI, and import fingerprints to my Sensor zfm-60. but seems i can't make it with this library, can you assist, pls?

```
root@pi:/usr/share/doc/python-fingerprint/examples# python3 example_downloadimage.py 
Currently used templates: 2/1000
Waiting for finger...
Downloading image (this take a while)...
Operation failed!
Exception message: image index out of range
```

philippmeisberger, 2019-01-08:
`downloadImage()` downloads the fingerprint image from sensor image buffer. Since the sensor stores only characteristics internally which are no image data you must use `readImage()` before calling `downloadImage()`.

I already saw this error in #50 ... I also got an ZFM-60 but cannot reproduce this on my machine (not tested on RPi)

ardawan, 2019-01-10:
> `downloadImage()` downloads the fingerprint image from sensor image buffer. Since the sensor stores only characteristics internally which are no image data you must use `readImage()` before calling [...]

which sensor you have tested and works perfectly? 
If the ZFM-60 got issue, i need to choose another sensor for my project. There is no time to waste on it.

ardawan, 2019-01-10:
> `downloadImage()` downloads the fingerprint image from sensor image buffer.

and that doesn't work at all in using ZFM-60 and this library. there is another library i could manage to extract the bitmap file data, but here no chance! as i mentioned the error.

some how, alternatively i get additional error messages! not sure what's that for 
```
root@pi:/usr/share/doc/python-fingerprint/examples# python3 example_downloadimage.py 
Currently used templates: 1/1000
Waiting for finger...
Downloading image (this take a while)...
downloading
Operation failed!
Exception message: image index out of range
Exception ignored in: <bound method PyFingerprint.__del__ of <pyfingerprint.pyfingerprint.PyFingerprint object at 0x76a3a970>>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pyfingerprint/pyfingerprint.py", line 165, in __del__
  File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 440, in close
TypeError: 'NoneType' object is not callable
```

ardawan, 2019-01-15:
> Cannot be reproduce on my Debian 9 64-bit using ZFM-60 with PL2303 USB-TTL converter:
> 
> ```
> python2 /usr/share/doc/python-fingerprint/examples/example_downloadimage.py
> Currently used templates: 1/1000
> Waiting for finger...
> Downloading image (this take a while)...
> The image was saved to "/tmp/fingerprint.bmp".
> ```

Yes i am using the same OS, same architecture and same sensor, but some how i won't get any result. I'm gonna buy another model of the sensor and check again.
Thanks for confirming

ardawan, 2019-01-21:
That is true, i already tested this library on R307 instead of ZFM-60 and this function works perfectly. Thanks for the quick replies.

closed 2019-01-21 by ardawan