bastianraschke/pyfingerprint #59: store data into the sensor in a differnet way - not like issue #42

ardawan, 2019-01-22:
i've done the same approach as #42 and it works. But i need to extract the data from the sensor and write into a file on my computer and then import to a new sensor via reading the extracted file/data.
to extract the data i did
```
position = 0
f.loadTemplate(0, 0x01)
characters = f.downloadCharacteristics(0x01)
w = open('./data', 'w')
w.write(str(characters))
w.close()
```
and i got 
```
[3, 1, 87, 0, 132, 0, 0, 04, 0, 04, 0, 04, 04, 126, 20, 0, 0, 14, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 2, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 36, 151, 87, 158, 56, 29, 86, 0, 28, 167, 108, 0, 82, 45, 43, 0, 78, 51, 106, 0, 28, 53, 213, 0, 9, 191, 40, 190, 36, 36, 86, 95, 94, 46, 20, 127, 94, 65, 39, 63, 0, 0, 171, 02, 44, 0, 142, 156, 34, 194, 19, 10, 12, 174, 0, 00, 9, 175, 0, 58, 10, 56, 0, 58, 88, 186, 211, 83, 41, 0, 170, 147, 14, 57, 216, 56, 63, 0, 20, 57, 93, 0, 147, 48, 116, 66, 0, 0, 58, 191, 79, 07, 103, 50, 148, 79, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 3, 1, 98, 17, 133, 0, 0, 04, 0, 04, 0, 04, 0, 30, 04, 6, 08, 2, 08, 0, 00, 0, 20, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 104, 22, 168, 04, 76, 34, 86, 0, 49, 44, 171, 190, 35, 51, 151, 126, 115, 52, 20, 04, 17, 55, 0, 190, 38, 63, 0, 0, 59, 28, 0, 159, 57, 41, 85, 0, 0, 176, 05, 95, 0, 56, 107, 159, 81, 59, 84, 31, 117, 0, 20, 183, 0, 184, 172, 0, 32, 167, 65, 186, 33, 42, 0, 91, 31, 187, 0, 01, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

in the next step i tried to read and convert to bytearray() and then upload to the sensor
```
r = open('./data', 'r')
b = bytearray(r.read(), 'utf-8')

f.uploadCharacteristics(0x01, b)
f.storeTemplate(100, 0x01)
```

Now i got registered into position 100 with no error.
But the problem is when i search for it, cannot match.

```
$ python3 /usr/share/doc/python-fingerprint/examples/example_search.py
Currently used templates: 1/1000
Waiting for finger...
No match found!
```

Is there some codes that i miss?

ardawan, 2019-01-22:
i found the issue was about my data. i didn't convert it into bytes properly. so after doing it properly, now it works with the same approach.

closed 2019-01-22 by ardawan

asifmp, 2022-08-26:
> i found the issue was about my data. i didn't convert it into bytes properly. so after doing it properly, now it works with the same approach.

how did you rectify?
please post