bastianraschke/pyfingerprint #61: Exporting and importing templates

Schaussi, 2019-02-05:
Hi,

thank your for writing this library, it's working great.
I am looking for the best way to keep multiple scanners' templates in sync.
In your blog you write it's possible to download and upload all templates.
Unfortunately, I have not found an example how to do this yet.
Could you provide me a how to read/store all templates of/to a scanner?

Thank you,
best regards

Schaussi, 2019-02-06:
Thank you, that worked perfectly.

closed 2019-02-06 by Schaussi

ashraful000, 2020-01-03:
@Schaussi could you explain how you did it. I passed characteristics as an argument in uploadCharacteristics(characteristics) then i called searchTemplate() but this gives me an error "The given charbuffer number is invalid!" @philippmeisberger

philippmeisberger, 2020-01-03:
@ashraful000 Remind that `uploadCharacteristics()` takes the char buffer ID (`0x01` or `0x02`) in the first argument

ashraful000, 2020-01-03:
@philippmeisberger Thank you so much.

bayutirto, 2020-01-04:
Hi, sorry can you write the code using uploadCharacteristics please? I still don't understand how to use it, I change downloadCharacteristics to uploadCharacteristics but it gives me an error, I already download the template but I don't know how to upload the template to another raspberry pi. Thank you!

Btw, is it possible to enroll the fingerprint template into mysql? I already use someone code but its upload the template position not the char buffer, I'm worry it can't be use for another rpi to access the fingerprint template. Thank you before :smile:

ashraful000, 2020-01-05:
@bayutirto if u want to upload template to the sensor just call the function `uploadCharacteristics(0x01,characteristics)`  , **characteristics** contains the template you want to upload to the sensor