bastianraschke/pyfingerprint #71: The received packet do not begin with a valid header! for Raspberry pi 3b +

jainal09, 2019-04-14:
Hi, I am using a UART to USB Converter for connecting the fingerprint sensor with my raspberry pi and below is the code which I have changed for my use.
https://gist.github.com/jainal09/db4074a4bd1107cc7f527ee51014855a

The Scripts runs fine for 4-5 times in infinite loop but after that, it pops up an exception 
The received packet do not begin with a valid header! 
The fingerprint sensor could not be initialized!

Re-Running the script fixes error up to 4-5 times.
Please Help!

jainal09, 2019-04-16:
Hi please help me its kind of like urgent as a whole project is depending on this thing!!

bastianraschke, 2019-04-17:
First, why you call the python code via the system command? Why not call direct your function or whatever - this does not make sense for me and adds unnecessary overhead.

It seems, the communication failed: Try to reduce the baudrate and/or use another converter and/or add some delay time after each loop. Also see FAQ #1 on https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/

bastianraschke, 2019-08-15:
I close the issue because of missing information updates.

closed 2019-08-15 by bastianraschke