bastianraschke/pyfingerprint #72: example_enroll.py not finding finger

Siddapa, 2019-04-21:
Once I run the .py file, my fingerprint sensor doesn't sense my finger. I have a FlashTree FMP12 and am aware that it isn't on the list of compatible sensors, but I saw a guide with this same sensor working with this module, Can anyone tell me what's wrong?

philippmeisberger, 2019-04-22:
Could you please provide the link to this guide? Do you have a datasheet? Please describe what you want to achieve and what you have tried yet.

Siddapa, 2019-04-24:
I figured out why it doesn't work. The sensor was built for an Arduino. This means that the raspberrypi can't read the data given by the Arduino. I found the correct GitHub repository for it, but it uses .ino files. Thx anyway

closed 2019-04-24 by Siddapa