bastianraschke/pyfingerprint #73: Unable to initialize fingerprint sensor

steelhayr, 2019-04-29:
Hi I am having some issue using the library when I try to connect a fingerprint sensor with an Arduino One board.

I want to try some examples, but I don't manage to make any of them work correctly. (I have tried the index and enroll examples)

I have tried to switch the green and the white wire, but I get the same error.

The error is the following one: The received packet do not begin with a valid header!

I am sure that the board and the sensor work correctly because I have tried them using Arduino IDE (I have executed several examples using the same board, sensor and wires without any error)

As far as I have reviewed, the error seems to happen when trying to close the connection at the __del__() method.

[Captura de pantalla -2019-04-29 15-35-19](https://user-images.githubusercontent.com/9446614/56899932-efa85000-6a94-11e9-9cb8-eb1e6beadc86.png)

I would like to know how I can find out which headers are valid? 

Thanks in advance. Any help is welcome.

bastianraschke, 2020-01-05:
You said, you tested the sensor with the Arduino on the pictures and it worked, but not with PyFingerprint? You tested the same methods? Same setup (baud rate, password etc.)? Which library and version did you used for the Arduino test?

steelhayr, 2020-01-05:
> You said, you tested the sensor with the Arduino on the pictures and it worked, but not with PyFingerprint? You tested the same methods? Same setup (baud rate, password etc.)? Which library and version did you used for the Arduino test?

Thank for answering, but I am not working in this project anymore, so you can close this issue.

closed 2020-01-07 by bastianraschke