bastianraschke/pyfingerprint #76: where it is stored and what format it is stored.

DOJUNKIM-YJ, 2019-07-19:
Hi i am using a fingerprint and one question.
When I register a fingerprint, I want to know where it is stored and what format it is stored. Is there any way I can see it? I'm a beginner, so thank you for the easy explanation!
Thank you!

philippmeisberger, 2019-07-19:
Hi. The fingerprint is just an image. When enrolling a finger the sensor extracts the characteristics from the taken image and stores it inside the internal database. Sadly the format is not documented anywhere. What we know: it has a length of 512 bytes.

philippmeisberger, 2019-07-19:
I forgot to say: You can download the characteristics by the `downloadCharacteristics()` method. Just take a look at the samples.

bastianraschke, 2019-08-15:
Also it is not an issue, so i close it.

closed 2019-08-15 by bastianraschke