bastianraschke/pyfingerprint #78: The received packet does not begin with a valid header! Raspberry Pi Zero

andim202, 2019-08-08:
Hi guys!

First of all thank you for your great guide, really appreciate it that you make the effort to accomplish such things.

I have problems whenever I input a valid finger. It initializes correctly but then stuck in a Error message loop: The received packet does not begin with a valid header!

I am using a Raspberry Pi Zero to ensure internet connectivity and SSHing to my PI. I really like to install my door control on our house. I added some parallelism to simultaneously look for any input from my code matrix and/or my fingerprint. Therefore I spawned 2 Threads which separately runs in a infinite loop. These are the components I have bought:

Fingerprint-Sensor:
https://www.amazon.de/gp/product/028492038X/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Relay:
https://www.amazon.de/gp/product/B00M8XJ71U/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Code Matrix:
https://www.amazon.de/gp/product/B00H8V49HY/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

Raspberry Pi:
https://www.amazon.de/gp/product/B06XCYGP27/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

TTL-Converter:
https://www.amazon.de/gp/product/B01MY038VE/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

And here I provide my code for any misleading actions I have made:

"""
DOOR PI
"""
import threading, time
from pyfingerprint.pyfingerprint import PyFingerprint
from keypad2 import keypad
import RPi.GPIO as GPIO

relay_open_time = 4
release_time = 10
code_sequence = [4, 1, 1, '1', '*']

GPIO.setmode(GPIO.BOARD)  # GPIO numbers instead of Board numbers
RELAIS_1_GPIO = 11
GPIO.setup(RELAIS_1_GPIO, GPIO.OUT)  # Apply GPIO mode
GPIO.output(RELAIS_1_GPIO, GPIO.HIGH)  # initially, turn off relay

GPIO.setwarnings(False)  # Disable warning to prevent interrupt

def open_door(message):
    print(message)
    GPIO.output(RELAIS_1_GPIO, GPIO.LOW)  # Relay opens door if code is valid
    time.sleep(relay_open_time)  # Open door for 4 seconds
    GPIO.output(RELAIS_1_GPIO, GPIO.HIGH)  # Relay closes door
    time.sleep(release_time)

def code_panel():
    kp = keypad(columnCount=4)
    while True:
        print("matrix")
        code_is_valid = True

        # Matrix has been initialized
        digit = None
        while digit is None:
            digit = kp.getKey()
            seq = []
            for i in range(5):
                digit = None
                while digit is None:
                    digit = kp.getKey()

                if code_is_valid:
                    seq.append(digit)
                    time.sleep(0.4)

                    # check sequence of code
                    print(seq)
                    if seq == code_sequence:
                        open_door("Code accepted!")
                else:
                    break
                if not code_is_valid:
                    break

def code_init():
    kp = keypad(columnCount=4)
    return kp

def fingerprint_exec(fingerprint):
    # Converts read image to characteristics and stores it in charbuffer 1
    fingerprint.convertImage(0x01)

    # Searches template
    result = fingerprint.searchTemplate()

    position_number = result[0]

    if position_number == -1:
        print('No match found!')

    else:
        open_door('Found template at position #' + str(position_number))

def finger_print():
    print("in finger print")
    fingerprint_sensor = None
    try:
        fingerprint_sensor = PyFingerprint('/dev/ttyUSB0', 57600, 0xFFFFFFFF, 0x00000000)

        if not fingerprint_sensor.verifyPassword():
            raise ValueError('The given fingerprint sensor password is wrong!')

    except Exception as e:
        print('The fingerprint sensor could not be initialized!')
        print('Exception message: ' + str(e))
        GPIO.cleanup()
        exit(1)

    # init_fingerprint()  # init only once
    while True:
        try:
            print('Waiting for finger...')
            while not fingerprint_sensor.readImage():
                pass  # If nothing has been found, just continue in loop
            fingerprint_exec(fingerprint_sensor)

        except Exception as e:
            print('Operation failed!')
            print('Exception message: ' + str(e))
            time.sleep(1)

if __name__ == '__main__':
    # lock = threading.Lock()

    fingerprint_thread = threading.Thread(target=finger_print, )
    matrix_thread = threading.Thread(target=code_panel, )

    # Spawn threads
    fingerprint_thread.start()
    matrix_thread.start()

    # Wait for threads to terminate
    matrix_thread.join()
    fingerprint_thread.join()

andim202, 2019-08-14:
Do you think it depends on my fingerprint model or some code-based failure?
Thank you guys!!

bastianraschke, 2019-08-15:
Hi andim202, did you check the FAQ regarding this problem: https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/ ?

On the Amazon picture, the sensor looks exactly like the ZFM-20, but you never can be sure if its a faulty clone of course.

Check if the sensor works with other Raspberry, without converter (directly on RPi, see Update 7). Also you could check if the sensor works with Adafruits Arduino Library (which we ported to Python and implemented most missing methods).

bastianraschke, 2020-01-05:
I close the issue because it seems to be a hardware issue.

closed 2020-01-05 by bastianraschke