bastianraschke/pyfingerprint #79: PyFingerprint.searchTemplate missing argument

raphaelreme, 2019-09-05:
Hi, 

I just used your lib with another device : r307. It works well, thanks a lot !

But I didn't agree with a small point in the method searchTemplate !

```python
def searchTemplate(self):
        """
        Search the finger characteristics in CharBuffer in database.
        Return a tuple that contain the following information:
        0: integer(2 bytes) The position number of found template.
        1: integer(2 bytes) The accuracy score of found template.
        @return tuple
        """

        ## CharBuffer1 and CharBuffer2 are the same in this case
        charBufferNumber = 0x01

        ## Begin search at index 0
        positionStart = 0x0000
        templatesCount = self.getStorageCapacity()

        packetPayload = (
            FINGERPRINT_SEARCHTEMPLATE,
            charBufferNumber,
            self.__rightShift(positionStart, 8),
            self.__rightShift(positionStart, 0),
            self.__rightShift(templatesCount, 8),
            self.__rightShift(templatesCount, 0),
        )

```
You use the command Search of the device which is for the r307 : 
```
Description: to search the whole finger library for the template that matches the one in CharBuffer1 or CharBuffer2. [..]
Input Parameter: BufferId, [...]
```

In fact I believe that your function should also take the bufferId as a parameter. Here It only searches a match with the 1st buffer. And one could possibly have put the finger of the user to identify in the second buffer, then he could not use your searchTemplate method.

I would propose to do smthg like : 

```python
def searchTemplate(self, charBufferNumber = 0x01):
[....]
```

Maybe it is only due to my specific device(r307)'s protocol, 

Raphael

philippmeisberger, 2019-10-22:
Hi @raphaelreme ,
you are right: It would be handy to change the charbuffer. I will implement your proposal ;)

philippmeisberger, 2019-10-22:
@bastianraschke Maybe we could add parameter `positionStart` to change the start position of the search. What do you think?

bastianraschke, 2019-10-31:
Hi @raphaelreme and @philippmeisberger, i totally agree with both of your suggestions.

bastianraschke, 2020-01-05:
@raphaelreme Change is implemented.

closed 2020-01-05 by bastianraschke