bastianraschke/pyfingerprint #89: searchTemplate() throwing "communication error" exception

ashraful000, 2020-01-05:
I am using two R307 sensor, one to use as database(where all the templates are stored) and other to get the scanned fingerprint's template.
i wrote a script named **find.py** which takes the scanned template and upload it to the other sensor so that i can search. So the very first run works just fine but after that no matter how many times i run the script it gives me **"communication error"** even if i unplug and plug it back. But if i run **example_search.py**  and  if it finds a match then **find.py** script works again for only once . _**I checked searchTemplate() was throwing this exception**_

bastianraschke, 2020-01-05:
Without code it is hard to get the issue. But it sounds like you do not handle the sensor right and it is in an illegal state to it denies to execute the command.

ashraful000, 2020-01-05:
@bastianraschke I also think i haven't handled it properly as i am just calling few  functions without any sort of understanding. 

**Here is the code :**

```
from pyfingerprint.pyfingerprint import PyFingerprint

try:
    f = PyFingerprint('COM6', 57600, 0xFFFFFFFF, 0x00000000)

    if ( f.verifyPassword() == False ):
        raise ValueError('The given fingerprint sensor password is wrong!')

except Exception as e:
    print('The fingerprint sensor could not be initialized!')
    print('Exception message: ' + str(e))
    exit(1)

try:
      
    print("1")
    chr =[3, 1, 82, 24, 130, 0, 255, 254, 255, 254, 255, 254, 255, 254, 128, 2, 0, 2, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 2, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 92, 22, 227, 126, 24, 32, 31, 222, 118, 163, 38,
    182, 49, 39, 219, 30, 16, 45, 92, 190, 35, 174, 27, 222, 116, 55, 105, 222, 53,
    191, 149, 94, 83, 24, 33, 95, 13, 29, 31, 223, 90, 30, 164, 223, 102, 158, 229,
    95, 37, 160, 159, 255, 77, 47, 169, 63, 83, 183, 169, 191, 39, 56, 24, 159, 103,
    56, 211, 63, 76, 187, 211, 223, 50, 33, 159, 26, 63, 177, 19, 242, 59, 177, 234
    , 114, 71, 149, 141, 15, 68, 20, 78, 240, 68, 156, 164, 42, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    , 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    , 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    , 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    , 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0,
    0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0
    , 0, 0, 0, 0, 0, 0]
    
    print("2")   
    f.uploadCharacteristics(0x01, chr)
    print("3")
    result = f.searchTemplate()
    print("4")

    positionNumber = result[0]
    accuracyScore = result[1]

    if ( positionNumber == -1 ):
        print('No match found!')
        exit(0)
    else:
        print('Found template at position #' + str(positionNumber))
        print('The accuracy score is: ' + str(accuracyScore))

    #f.loadTemplate(1, 0x01)
    #f.downloadCharacteristics(0x01)
    #chr = [0]
    #f.uploadCharacteristics(0x01, chr) 
    #print(f.downloadCharacteristics(0x01))   

   

except Exception as e:
    print('Operation failed!')
    print('Exception message: ' + str(e))
    exit(1)

```

bastianraschke, 2020-01-07:
After the upload of the characteristics you need create a template (`storeTemplate`) of them to be able to search for it. This may be the issue.

derit, 2020-03-31:
@bastianraschke think this case for search if template is exist if not exist storeTemplate