bastianraschke/pyfingerprint #90: Problems with the R503 Sensor

Jako0815, 2020-01-11:
Hello,
I have problems with the R503 Sensor, when i tried to enroll a new finger, the script always says that the 2 fingers do not match. I chanced a few lines of code and it worked fine for me. And I like to share my script, if you have the same problems.
Here is it:
`#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
PyFingerprint
Copyright (C) 2015 Bastian Raschke <bastian.raschke@posteo.de>
All rights reserved.

@author: Bastian Raschke <bastian.raschke@posteo.de>
"""
import hashlib
import time
from pyfingerprint.pyfingerprint import PyFingerprint

print('Fingerprint Script')

## Enrolls new finger
##

## Tries to initialize the sensor
try:
    f = PyFingerprint('/dev/ttyUSB0', 57600, 0xFFFFFFFF, 0x00000000)

    if ( f.verifyPassword() == False ):
        raise ValueError('The given fingerprint sensor password is wrong!')

except Exception as e:
    print('The fingerprint sensor could not be initialized!')
    print('Exception message: ' + str(e))
    exit(1)

## Gets some sensor information
print('Currently stored templates: ' + str(f.getTemplateCount()))

## Tries to enroll new finger
try:
    print('Waiting for finger...')

    ## Wait that finger is read
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 1
    f.convertImage(0x01)
    f.createTemplate()

    ## Checks if finger is already enrolled
    result = f.searchTemplate()
    positionNumber = result[0]

    if ( positionNumber >= 0 ):
        print('Template already exists at position #' + str(positionNumber))
        exit(0)

    print('Remove finger...')
    time.sleep(2)

    print('Waiting for same finger again...')

    ## Wait that finger is read again
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 2
    f.convertImage(0x02)
    f.createTemplate()

    if ( f.compareCharacteristics() == 0 ):
      print(str(f.compareCharacteristics()))
      print('Fingers do not match')
      exit(0)

    ## Compares the charbuffers and creates a template
    f.createTemplate()

    ## Gets new position number (the counting starts at 0, so we do not need to increment)
    positionNumber = f.getTemplateCount()

    ## Saves template at new position number
    if ( f.storeTemplate(positionNumber) == False ):
        print('Finger enrolled successfully!')
        print('New template position #' + str(positionNumber))
    print('Waiting for finger...')
except Exception as e:
    print('Operation failed!')
    print('Exception message: ' + str(e))
    exit(1)#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
PyFingerprint
Copyright (C) 2015 Bastian Raschke <bastian.raschke@posteo.de>
All rights reserved.

@author: Bastian Raschke <bastian.raschke@posteo.de>
"""
import hashlib
import time
from pyfingerprint.pyfingerprint import PyFingerprint

print('Fingerprint Script')

## Enrolls new finger
##

## Tries to initialize the sensor
try:
    f = PyFingerprint('/dev/ttyUSB0', 57600, 0xFFFFFFFF, 0x00000000)

    if ( f.verifyPassword() == False ):
        raise ValueError('The given fingerprint sensor password is wrong!')

except Exception as e:
    print('The fingerprint sensor could not be initialized!')
    print('Exception message: ' + str(e))
    exit(1)

## Gets some sensor information
print('Currently stored templates: ' + str(f.getTemplateCount()))

## Tries to enroll new finger
try:
    print('Waiting for finger...')

    ## Wait that finger is read
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 1
    f.convertImage(0x01)
    f.createTemplate()

    ## Checks if finger is already enrolled
    result = f.searchTemplate()
    positionNumber = result[0]

    if ( positionNumber >= 0 ):
        print('Template already exists at position #' + str(positionNumber))
        exit(0)

    print('Remove finger...')
    time.sleep(2)

    print('Waiting for same finger again...')

    ## Wait that finger is read again
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 2
    f.convertImage(0x02)
    f.createTemplate()

    if ( f.compareCharacteristics() == 0 ):
      print(str(f.compareCharacteristics()))
      print('Fingers do not match')
      exit(0)

    ## Compares the charbuffers and creates a template
    f.createTemplate()

    ## Gets new position number (the counting starts at 0, so we do not need to increment)
    positionNumber = f.getTemplateCount()

    ## Saves template at new position number
    if ( f.storeTemplate(positionNumber) == False ):
        print('Finger enrolled successfully!')
        print('New template position #' + str(positionNumber))
    print('Waiting for finger...')
except Exception as e:
    print('Operation failed!')
    print('Exception message: ' + str(e))
    exit(1)`

chrisb2, 2020-04-02:
This also worked for me with the R503 sensor. Replicated above program here in readable format with a comment on the 2 lines added:
```
#!/usr/bin/env python
# -*- coding: utf-8 -*-

"""
PyFingerprint
Copyright (C) 2015 Bastian Raschke <bastian.raschke@posteo.de>
All rights reserved.
"""

import time
from pyfingerprint.pyfingerprint import PyFingerprint

## Enrolls new finger
##

## Tries to initialize the sensor
try:
    f = PyFingerprint('/dev/ttyUSB0', 57600, 0xFFFFFFFF, 0x00000000)

    if ( f.verifyPassword() == False ):
        raise ValueError('The given fingerprint sensor password is wrong!')

except Exception as e:
    print('The fingerprint sensor could not be initialized!')
    print('Exception message: ' + str(e))
    exit(1)

## Gets some sensor information
print('Currently used templates: ' + str(f.getTemplateCount()) +'/'+ str(f.getStorageCapacity()))

## Tries to enroll new finger
try:
    print('Waiting for finger...')

    ## Wait that finger is read
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 1
    f.convertImage(0x01)
    f.createTemplate()  # Line added for fix

    ## Checks if finger is already enrolled
    result = f.searchTemplate()
    positionNumber = result[0]

    if ( positionNumber >= 0 ):
        print('Template already exists at position #' + str(positionNumber))
        exit(0)

    print('Remove finger...')
    time.sleep(2)

    print('Waiting for same finger again...')

    ## Wait that finger is read again
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 2
    f.convertImage(0x02)
    f.createTemplate()  # Line added for fix

    ## Compares the charbuffers
    if ( f.compareCharacteristics() == 0 ):
        raise Exception('Fingers do not match')

    ## Creates a template
    f.createTemplate()

    ## Saves template at new position number
    positionNumber = f.storeTemplate()
    print('Finger enrolled successfully!')
    print('New template position #' + str(positionNumber))

except Exception as e:
    print('Operation failed!')
    print('Exception message: ' + str(e))
    exit(1)
```

chrisb2, 2020-04-27:
I have been experimenting with this and  believe a better solution is to use "create template" instead of "compare characteristics", this is based on the data sheet which says for "create template":

>To combine information of character files from CharBuffer1 and CharBuffer2 and
generate a template which is stored back in both CharBuffer1 and CharBuffer2.

and:

>Confirmation code=0aH: fail to combine the character files. That’s, the character files
don’t belong to one finger.

So your program would become:
```
## Tries to enroll new finger
try:
    print('Waiting for finger...')

    ## Wait that finger is read
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 1
    f.convertImage(0x01)
  
    ## Checks if finger is already enrolled
    result = f.searchTemplate()
    positionNumber = result[0]

    if ( positionNumber >= 0 ):
        print('Template already exists at position #' + str(positionNumber))
        exit(0)

    print('Remove finger...')
    time.sleep(2)

    print('Waiting for same finger again...')

    ## Wait that finger is read again
    while ( f.readImage() == False ):
        pass

    ## Converts read image to characteristics and stores it in charbuffer 2
    f.convertImage(0x02)
  
    ## Compares the charbuffers and creates template if they are the same finger
    if ( not f.createTemplate() ):
        raise Exception('Fingers do not match')

    ## Saves template at new position number
    positionNumber = f.storeTemplate()
    print('Finger enrolled successfully!')
    print('New template position #' + str(positionNumber))
```