bastianraschke/pyfingerprint #91: Cant build package

kostek443, 2020-02-08:
I want build package on debian 10.
When i use command `dpkg-buildpackage -uc -us
`
I get this error
`unning install_scripts
   dh_install -O--buildsystem=pybuild
   dh_installdocs -O--buildsystem=pybuild
   dh_sphinxdoc -O--buildsystem=pybuild
dh_sphinxdoc: error: unknown JavaScript code: debian/python-fingerprint-doc/usr/share/doc/python-fingerprint-doc/html/_static/doctools.js
make: *** [debian/rules:6: binary] error 255
dpkg-buildpackage: error: debian/rules binary subprocess returned exit status 2`

How can i resolve this ?

philippmeisberger, 2020-02-09:
I cannot reproduce this on my Debian 10. There must be something special with your Debian or we forgot to mention a package dependency. Did you exactly follow the build instructions?

BTW: I am hosting an APT repository with the package: https://www.pm-codeworks.de/pages/apt-repository-en.html

kostek443, 2020-02-10:
I just download your repository , install by pip your library and now it's working.

closed 2020-02-11 by kostek443