bastianraschke/pyfingerprint #92: uploadCharacteristics function not working properly

ivanbazina, 2020-02-26:
uploadCharacteristics function doesn't work (Exception message: communication error).
I downloaded characteristics of fingerprint with downloadCharacteristics function and stored it on my PC. Then deleted that fingerprint and tried to upload it back.

- I suspect that it is because of data length which is 512 and I am not shure if it should be sent in 2 or 3 separate packages (because maxPackageLength is 256, and checksum is also calculated in packageLength)

- Also I saw in sensor documentation this line: "Data packet; Data packet shall not appear alone in executing processs, must follow command packet or acknowledge packet."
I don't see any acknowledge packets sent after data packets.

Does anyone have similar problems?
I will try to fix this but I just started using lib few days ago and if you can help with any idea what is wrong it would be nice.

philippmeisberger, 2020-02-28:
Which sensor do you use? Could you provide some of your code?

ivanbazina, 2020-02-28:
Snesor is R307.
Code is cloned from this repo on master branch.

Update: After I changed maxPacketSize variable in uploadCharacteristics function to 32 it works. (256 is what I get from module, but after that I overwrite it to 32 and then it works) .

Also I added one more line of code for fixing the packets which have length that is not multiple of 32.

I will create pull request so you can see the code.

ivanbazina, 2020-02-28:
I just tried to push new branch with my fixes and I don't have permissions to do it.
Do I need to fork repository before or what?

philippmeisberger, 2020-02-28:
Sure, you need to fork this repo and create a pull request

ivanbazina, 2020-03-02:
I created 2 pull requests: 
#93 - I think it should be done anyway
#94 - As I told you, uploading characteristics didn't work for me until I hardcoded maxPacketLength to 32 even it is set to 256 on device.

There is also 1 question in #94 because I am not sure about it - for me it works without that code.