bastianraschke/pyfingerprint #96: readPacket function does not handle case were no byte is read

chrisb2, 2020-04-02:
At [L315](https://github.com/bastianraschke/pyfingerprint/blob/Development/src/files/pyfingerprint/pyfingerprint.py#L315) in the __readPacket() function a check is made that a byte has been received, but the case where it not received is not handled. I believe the code should look like this:
```
            if ( len(receivedFragment) != 0 ):
                receivedFragment = self.__stringToByte(receivedFragment)
                ## print 'Received packet fragment = ' + hex(receivedFragment)
            else:
                continue
```
Happy to raise a PR for this is you like.

regards,
Chris

philippmeisberger, 2020-04-27:
Hi @chrisb2 ,
this could may fix a really annoying bug which we are struggling for years ;) I will create a fix for that!

philippmeisberger, 2020-06-25:
Should be fixed. Therefore I close this issue.

closed 2020-06-25 by philippmeisberger