bastianraschke/pyfingerprint #97: exit status 2 error

snappercayt, 2020-04-19:
Extension error:
Unknown event name: autodoc-process-docstring
debian/rules:15: recipe for target 'override_dh_auto_build' failed
make[1]: *** [override_dh_auto_build] Error 1
make[1]: Leaving directory '/home/pi/pyfingerprint/src'
debian/rules:6: recipe for target 'build' failed
make: *** [build] Error 2
dpkg-buildpackage: error: debian/rules build gave error exit status 2

philippmeisberger, 2020-04-25:
It seems that there is a missing dependency... Hard to guess which is missing.
Did you install all the required build dependencies?

snappercayt, 2020-04-25:
I did run the commands in the sequence it was given, I hope all the dependencies were Installed like that.
If not then how do I find which ones are missing?

philippmeisberger, 2020-04-27:
OK. Hard to guess which are missing... 
Anyway, I am hosting an APT repository which contains pre build PyFingerprint Debian package. Here is the setup how to add it: https://www.pm-codeworks.de/pages/apt-repository-en.html.

After that you can install PyFingerprint by the this command: `apt install python3-pyfingerprint`

snappercayt, 2020-04-28:
that is interesting thank you so much, that would be helpful for those who want to start it with that

closed 2020-04-28 by snappercayt