bastianraschke/pyfingerprint #99: Exception message: The received packet do not begin with a valid header!

snappercayt, 2020-05-11:
Hi, 
made some changes to the code to work on Raspberrypi 3b+ on its serial port
changed ttyUSB0 to ttyS0 so now GPIO should be working as the port for the sensor 
but this is not working
```
def __init__(self, port = '/dev/ttyS0', baudRate = 57600, address = 0xFFFFFFFF, password = 0x00000000):
 ```
I did not change anything else so what should be giving this error?

philippmeisberger, 2020-05-11:
Hi,
please see https://sicherheitskritisch.de/2015/03/fingerprint-sensor-fuer-den-raspberry-pi-und-debian-linux-en/, #1 or #71.

snappercayt, 2020-05-12:
Thanks! I did check that and what not lol tried to edit the read header function in the PyFingerprint 
then went back to the same file, I know its the old one because I removed all the files and installed it again and now I see this
```
>>> %Run enroll.py
The fingerprint sensor could not be initialized!
Exception message: 'module' object is not callable
```

matigarowel, 2022-01-01:
I solve this using this command **sudo stty -F /dev/ttyS0 57600**